spacecraft-design-lab-2019/documentation #21: Conceptual battery & board packaging in internal satellite volume

Jeff-Robinson, 2019-10-04:
2- and 3-board layouts; figure out max possible battery capacity and potential cell selection & layout options

maholli, 2019-10-19:
The various one-on-one meetings on 10/17 further refined some implementation details of the spacecraft. Below is the resulting whiteboard sketches, but since there are inconsistencies (as thoughts evolved) I will post some summary sketches as well. 
[image]

maholli, 2019-10-19:
### Summary sketch

[image]

### Camera view

[image]