spacecraft-design-lab-2019/documentation #68: Power Board Testing

thwhite, 2019-10-15:
Steps:
- Attach 3.3V and GND to the power input w/ power supply. 
- Confirm that VBAT-OK goes to 3.3V
- If yes, put the jumper on, and measure the output of JP3. Should be 3.3V. 
- Then hook up a solar cell. Minimum voltage? Measure the voltage the cell generates normally. 
- Measure the current between the external battery pins. 
- Note voltage of battery disconnected. 
- Stretch goal: measure voltage between solar cell inputs. Should change that to meet its needs.

maholli, 2019-10-17:
NOTE: make sure R7 is populated. Otherwise you'll need to use a jumperwire between the two pins shown below in red.
[image]

maholli, 2019-10-17:
End goals:

- [ ] using a power supply as an input on the battery input terminals (MAX 4.5V, try 3.3V), can you make the board generate 3.3V as measured between the points pictured below?
   - [ ] how low can the power supply go before 3.3V is no longer generated?
- [ ] applying the power supply to the solar input (MAX 5.5V, try 5V), are you able to charge a single cell LiPo battery? What voltage does charging stop/start?
   - [ ] what is the maximum charge current into the battery?

reference image (since I did a poor job of labeling the board 🙂)
[image]

maholli, 2019-10-21:
Did cursory checks and confirmed board is working as intended. 

- [ ] Need to do in-depth measurements and explore parallel VOUT schemes. 

New issue for parallel output design: #81

maholli, 2019-10-30:
note: @zchen131 has detailed notes documenting her characterization work in https://github.com/spacecraft-design-lab-2019/SolarActuators/tree/master/EnergyHarvesterBoardTest

### 2019-10-29 Next Steps
Based on Zhengyu's recent observations, it appears the following is occurring... Two EHB-10 boards with separate solar inputs have their outputs tied together (3V3 and GND). In the lab, the cells are ~1.2V. The voltage measured between 3V3 and GND is properly regulated to 3.3V until a 165Ohm resistor is connected between 3V3 and GND to simulate a load. Thereafter, the regulated voltage drops to nominal 0V. **This is likely because the solar cells are not able to sustain the power generation necessary and the regulator drops out.** Things to try next:

- [ ] using the exact same setup, measure and confirm 3V3 to GND is still ~0 to 0.2V, THEN remove the potentiometer from the breadboard and confirm that the 3V3 wire returns to 3.3V. This will demonstrate that the circuit is able to recover on its own and that it isn't stuck in a drop-out state.
- [ ] take your setup outside into direct sunlight. Wear sunglasses :dark_sunglasses: because the reflection off the cells can hurt your eyes! Then make the same measurements. 

It's possible the second experiment above will still not be enough power generation (depending on the cells, available sunlight, etc..). We can simulate the solar cells by using a bench-top power supply, but we need to be careful since the circuit is expecting the high-impedance input of a solar cell. Therefore, if a power supply is used, a voltage of ~3V needs to be used so the circuit is able to properly boost the input as it's expecting

zchen131, 2019-10-30:
[Energy_harvesting_board_testing_1026.pdf](https://github.com/spacecraft-design-lab-2019/documentation/files/3790896/Energy_harvesting_board_testing_1026.pdf)
[Energy_harvesting_board_testing_1029.pdf](https://github.com/spacecraft-design-lab-2019/documentation/files/3790897/Energy_harvesting_board_testing_1029.pdf)
[Energy_harvesting_board_testing_1030.pdf](https://github.com/spacecraft-design-lab-2019/documentation/files/3790898/Energy_harvesting_board_testing_1030.pdf)

maholli, 2019-10-30:
Congrats!!! 🎉🎉 

From your 10-30 notes it looks like the 3V3 output is steady (for your given sunlight/configuration) until more than ~30mA of current is drawn. This is not surprising and likely because that's the most the solar cells can produce for the given conditions. 

Next steps:

- [ ] attach the battery to the EXT BAT connectors on **both** EHB boards (aka in parallel). then perform the same sunlight experiment as conducted on 10-30. measure battery voltage before connecting, while connected, and again after ~5 minutes in the sun. 
- [ ] after performing the above 👆 experiment, duplicate the experiment while measuring the current flow into/out-of the battery. Find the Fluke quick-start manual in the cabinet for instructions on how to properly measure current with the multimeter. 
- [ ] separate from the two experiments above, I'd like to see the behavior of 3V3 if we put a schottky barrier diode on each 3V3 line before connecting it to the potentiometer. I'll need to find an appropriate diode for you to use, and there will be a voltage drop (multimeter will read less than 3.3V), but I want to see if the 3V3 line becomes more or less stable with the addition of the diodes.

### in general, STOP any experiments if the circuit or the battery start to feel hot 🔥 or the battery starts to swell.

zchen131, 2019-11-05:
Battery charging test is performed, the charging current is around 20-28 mA and we are able to get 0.1 V increase with 5mins charging. 
Reference documents can be found below:
EnergyHarvesterBoardTest/Energy harvesting board testing 1105.pdf