spacecraft-design-lab-2019/documentation #105: Characterize Solar Cells

klwall423, 2019-10-28:


maholli, 2019-10-29:
All characterization stuff should go in repo: https://github.com/spacecraft-design-lab-2019/solar-characterization

We're building and using this IV Swinger board: https://www.instructables.com/id/IV-Swinger-2-PCB-PV-Module-SSR/

- read https://github.com/spacecraft-design-lab-2019/solar-characterization/blob/master/ChrisEmail.txt then look at the [spreadsheet Chris made for us](https://github.com/spacecraft-design-lab-2019/solar-characterization/tree/master/IVS2_Scaling.xlsx) to select the proper resistor and capacitor values
- put together a Digi-Key cart with the components that we need to build the IV Swinger. Make sure component package, value, and quantities are appropriate

klwall423, 2019-11-01:
https://github.com/spacecraft-design-lab-2019/solar-characterization/blob/master/IV_Swinger_2_BOM.xlsx - bill of materials for IV_Swinger (Amazon and Digikey carts are linked in spreadsheet)