spacecraft-design-lab-2019/documentation #115: Primary flex cable design

maholli, 2019-10-30:
Need to architect the 24-pin flex cable that runs from the mainboard to all 6 external panel boards.

maholli, 2019-10-30:
beginnings of a pinout sketch:
[image]

maholli, 2019-10-30:
## Images to help with conceptualizing cable flow.

⭐red arrow points in the direction of cable insertion and denotes the side of the connector where pin 1 resides. Connector contacts on BOTH sides of the cable, so as long as the signals are in the right order, it doesn't matter whether they're on the top or the bottom of the cable when it reaches the connector.

- [connector datasheet](https://omronfs.omron.com/en_US/ecb/products/pdf/en-xf3m_xf2m.pdf)

[image]

[image]

[image]

maholli, 2019-10-30:
My attempt at sketching what a folded cable might look like: 
[image]

JTBee1000, 2019-11-01:
My Todos:
- [ ] Download Fusion 360
- [ ] Determine pad location for every connection

JTBee1000, 2019-11-01:
Note: SDA1 and SDA2 need to go to any three boards each.

JTBee1000, 2019-11-04:
Initial flex cable design--most direct
https://www.lucidchart.com/invitations/accept/b0606187-e3a2-40b4-9e76-91db044b9bdf

maholli, 2019-11-05:
Great start- having it in lucidchart will make the next step more straight forward as well. Next step should be working with someone in mechanical to orient the connectors for each panel accurately relative to each other. Imagine "unfolding" the entire satellite and overlaying each connector in the proper orientation and position in lucidchart

JTBee1000, 2019-11-05:
@ElliseD would love to sit down and talk about flattening out the cuboid CAD so I can think about how to route cables :)

maholli, 2019-11-05:
@JTBee1000 

Example of **one possible** cable routing scheme (available in cables repo). There are lots of other possible permutations that may be better. 

Some thoughts:

- Ground reference certainly isn't ideal, but it's not horrible.
- I'm open to rearranging motherboard connector as well as panel connectors, but we have to ensure it's exactly how we want it and everything routes properly before we make that call.

Two panel connectors:
[image]

close-up of the first interchange:
[image]

close-up of the second interchange:
[image]

JTBee1000, 2019-11-13:
@maholli how possible would it be to move the Z face (not sure +/-, the one nearest the motherboard pin out) all the way up so I don't have to transit through the mother and camera boards? Would necessarily have to make cutouts.

JTBee1000, 2019-11-13:
@maholli Ok, I've finished a first spin. I don't yet know where the cable on the top side (with antenna) go, so haven't done it yet. I'm also confused for the bottom face.... how am I supposed to get around the camera if there is insufficient width around the lens? Perhaps it makes more sense to rotate the camera so there is no blockage and I can come down with that same cable from the same size as the MB output.

maholli, 2019-11-13:
> @maholli Ok, I've finished a first spin. I don't yet know where the cable on the top side (with antenna) go, so haven't done it yet. I'm also confused for the bottom face.... how am I supposed to [...]

Ping @thwhite on status of CAD output for +Y board. I marked its orientation and pin1 location on the snapshots in https://github.com/spacecraft-design-lab-2019/documentation/issues/115#issuecomment-547994571.

As for making it around the camera board, there should be a small cutout that allows the cable to travel in the -Y direction. See my original sketch for what I'm imagining that'd look like. It can also travel along the lens- no issue there.

JTBee1000, 2019-11-13:
> Ping @thwhite on status of CAD output for +Y board. I marked its orientation and pin1 location on the snapshots in [#115 (comment)](https://github.com/spacecraft-design-lab-2019/documentation/issues/115#issuecomment-547994571).

Ok, I've asked him. 

> As for making it around the camera board, there should be a small cutout that allows the cable to travel in the -Y direction. See my original sketch for what I'm imagining that'd look like. It can also travel along the lens- no issue there.

I'm not sure what you mean about this. Can we chat in person today about this?

JTBee1000, 2019-11-17:
Ok, done with first complete pass of the flex cable design that works in 3D.

JTBee1000, 2019-11-17:
Now finished with (manually) creating a flattened model of the flex cable so that trace routing can begin. Exporting to DXF for Max.

JTBee1000, 2019-11-17:
Ok, @maholli here is the DXF. May want to check it for readability etc. Git won't let me upload that file type as a comment, so it's in the "cables" repository. Spent all morning on this, now need to dedicate time to other classes.

maholli, 2019-11-17:
@JTBee1000 the dxf file is not correct. It imports into KiCad as a line:
[image]

Please fix the dxf as well as route the cable as we had discussed.

JTBee1000, 2019-11-18:
Hm. Ok. I just selected Export in the Fusion 360 Menu and selected DXF. Here's a screenshot of what it looks like. I'm not sure how else to export to DXF except using Fusion 360. 
[image]

maholli, 2019-11-18:
In the navigation tree to the left of the viewport, right click on the particular **sketch** you want to export and then select DXF. I also have step-by-step for exporting a DXF of a 3D face as described starting here: https://spacecraftdesignlab.slack.com/archives/CNDR3K999/p1572324860020600

maholli, 2019-11-18:
PDF output of cables kicad schematic:

[FFC24.pdf](https://github.com/spacecraft-design-lab-2019/documentation/files/3861188/FFC24.pdf)

maholli, 2019-11-19:
RE initial routing attempt on lucid: 
You need to route all the signals at a time so you know what needs to be continued, etc... Screenshot below, you can see that signals not routed (like COIL4) are now pinched off. Take a look at my example above (https://github.com/spacecraft-design-lab-2019/documentation/issues/115#issuecomment-549951536) again.

[image]

JTBee1000, 2019-11-21:
hey @maholli just took another turn on it--see screenshot. Feedback appreciated. 
[image]

maholli, 2019-11-21:
@JTBee1000 yes! you got it.

JTBee1000, 2019-11-22:
Ok. I will try to finish the other two tonight.

JTBee1000, 2019-11-22:
@maholli I think it's done! Would love your detailed thoughts. One question, the Batt Stat never goes anywhere. This is intentional?
[image]

JTBee1000, 2019-11-22:
please look in detail via the lucidchart link

JTBee1000, 2019-11-23:
hi @maholli , if I'm correct, only thing left to do will be for you to recreate what I made in Kicad?

maholli, 2019-11-24:
@JTBee1000 let me take a crack at it and let you know how it goes

maholli, 2019-11-24:
@JTBee1000 where is the most recent DXF of the cable?

JTBee1000, 2019-12-02:
@maholli uploaded DXF to cables repo. Under name "dxf flatflex 1.0 for max.dxf"

maholli, 2019-12-02:
That DXF works- thanks!

maholli, 2019-12-02:
Posting my findings for flat flex cable dimensions:
[image]

We're using 
1. XF2M-2415-1A
2. XF2M-1015-1A 
[image]

maholli, 2019-12-04:
Layout of cablev00 finished:
[image]