spacecraft-design-lab-2019/documentation #116: kit mainboard

maholli, 2019-10-31:
Use the Interactive Bill-of-Materials section on the wiki: https://github.com/spacecraft-design-lab-2019/documentation/wiki to determine what parts are on-hand. 

📅 Complete BEFORE Tuesday 11/05.

- sort the parts into your team's respective bin
- parts are located in RExLab (Durand 032)
- the workstation is pictured below
- shared parts can be placed in either team's bin
- organize resistors, capacitors, and everything else into separate piles

[image]

NickG-NZ, 2019-11-01:
Kitted the OpenMV, and made a list of components that aren't in the boxes.

Known:
--------
LED's (Tricolor and VSMY12850)
STM32H7
OV7725
Pins

Other:
-------
C1, C4 -> 1uF 0603-CAP
C48, C49 -> 2.2uF 0603-CAP
C18 -> 4.7uF 0603-CAP

R2, R3, R9 -> 10K 0603-RES
R1 -> 10R 0603-RES
R5 -> 100K 0603-RES

L1, L2 -> MPZ1608S181A 0603
RN2, RN3 -> 10K EXBV8V

C11, C12, C21: 47uF EIA3216 -> have 47uF 1206 (metric vs imperial compatible?)

JeffersonAero, 2019-11-04:
Kitted the Mainboard. Components not found in boxes are documented below.

Found:
--------
C1, C4, C8, C9, C12
C2, C7, C10, C14
C11, C13
D2
R2, R3, R4, R5, R6, R7, R8
U1, U8
U3, U5
J3
J5
J4

Not Found:
-------
C3, C5, C6 -> 1uF
R1 -> 330 ohm
L1 -> 10uH
D1 -> LED_0603_1608
U2 -> ATSAMD51G_TQFN48
U6 -> MAX708RESA-T
U4 -> TPS82740BSIPR (may have this? Couldn't tell. There was another component with the TPS preface)
X1, X2 -> Antenna_U.FL (U.FL_Molex_MCRF_73412-0110_Vertical)
SW1 -> KMR231NGLFS
IC1 -> BMX160
J2 -> 10118194-0001LF
J1, J28, J29, J30 -> Conn_01x01

maholli, 2019-11-05:
all accounted for

closed 2019-11-05 by maholli