spacecraft-design-lab-2019/documentation #117: kit mainboard

maholli, 2019-10-31:
Use the Interactive Bill-of-Materials section on the wiki: https://github.com/spacecraft-design-lab-2019/documentation/wiki to determine what parts are on-hand. 

📅 Complete BEFORE Tuesday 11/05.

- sort the parts into your team's respective bin
- parts are located in RExLab (Durand 032)
- the workstation is pictured below
- shared parts can be placed in either team's bin
- organize resistors, capacitors, and everything else into separate piles

[image]

## Boards to Kit:

- [ ] +/- x panel
- [ ] +/- y panel
- [ ] +/- z panel

klwall423, 2019-11-05:
Z:

Not Found:
C1 - 1.0 uF capacitor
C6,11,12,14 - 100 uF capacitor
R1,2,3,4,14-19 - 0M resistor
R8,10 - 8.25M resistor
R3 - 10k resistor
R4 - 4.7k resistor
R5 - 100k resistor
R5 - 7.32M resistor
D2,3 - MSS1P4-M3-89A
U1 - NDS8434
JP1 - Jumper
SC1-8 - solar cells
Q1 - IRLML2803

-Y:

Not Found:
(not including parts in Z)
Battery connector

X:

Not Found:
(not including parts in -Y/Z)

maholli, 2019-11-05:
all accounted for

closed 2019-11-05 by maholli