spacecraft-design-lab-2019/documentation #132: Evaluate Panel Functionality

maholli, 2019-11-09:
Now that boards are built, it's time to check if they function! Primary goal is deciding whether or not your board functions as intended. 

## Primary Criteria 

- [ ] Does your board generate the expected open-circuit voltage (Voc)? Keep in mind there is the energy harvesting MPPT connected to the solar cells. How does the MPPT effect your Voc reading?
- [ ] If your board has an antenna connector, coordinate with @Timvrakas to determine antenna functionality. 
- [ ] If your board has a burn wire circuit, nichrome wire will need to be soldered into place and a burn-wire test scheme developed
- [ ] If you board has kill switches, use the continuity mode on a multimeter to confirm the high-side and low-side switches are working as intended. 
- [ ] Torque coils will need 0ohm resistors installed prior to testing. Resistors should be installed once a cohesive coil-winding pattern is decided.
- [ ] Sun sensor verification will require getting up-to-speed with circuitpython and leveraging the existing circuitpython libraries for the TSL2561. See https://learn.adafruit.com/tsl2561 to get started. Additionally, a 10-pin FFC breakout cable will be necessary to interface with the I2C signals.

## Energy Harvesting Circuit

NOTE: ❗ The energy harvesting circuit should be evaluated following the same series of checks as documented in #68. 

- [ ] Is the energy harvesting circuit generating a 3.3V regulated output? Note there is a diode on the 3.3V output, so measure before and after the diode. 
- [ ] Attach a LiPo battery to the board (10-pin FFC to LiPo connector necessary), does the circuit behave as expected? (see #68)

thwhite, 2019-11-11:
Getting started with the power testing. Would be useful to make a harness to break out the connector so we can easily access all those pins.

thwhite, 2019-11-11:
Contacted Tim to set up a time to verify antennas

thwhite, 2019-11-11:
Burn wire test scheme: 
1. Solder on burn wire to relevant contact. 
2. Connect the flat flex breakout (see above) to 1. 
3. Apply power to the digital pin turning on and off the burn wire. 
4. Observe: does the wire melt?

Big question: is the best way to control actuation through the mainboard? Or can we jank something together?

thwhite, 2019-11-11:
Voltage across one solar cell currently 0.7V at max sun lamp, with nothing connected.

zchen131, 2019-11-12:
Energy harvest circuit is tested under the lamplight, the report can be found using the link below:
https://github.com/spacecraft-design-lab-2019/SolarActuators/blob/master/EnergyHarvesterBoardTest/Energy_harvesting_board_testing_1111.pdf

A short summary is:
1. Lamplight cannot generate enough power as natural sunlight did. Dropped around 13% - 22%
2. Connect diode should increase the 3.3V from 0.3-0.4 to 0.7-0.8V in the lamplight scenario.

thwhite, 2019-11-12:
BIG DEBUGGING COMMENT for Solar Power:

First mystery: all the solar cells in line produce 0.5V, which is manifestly incorrect. They should be producing more than that. 

POSSIBLE THEORIES:
- Could be solar cells installed backwards. 
- Could be solar cells designed backwards. 
- Could be big losses across diodes - could solve with voltage check of them. 
- Could be a backwards diode somewhere is ruining everything - could solve by probing diodes. 
- Our contact with the cell is bad. (Can resolve this by soldering wires on - Max will do). 
- The light is just not strong enough. 

FURTHER EVIDENCE:
- The bypass diodes each have 0.7V across them.

zchen131, 2019-11-12:
The energy harvest board is tested under sunlight. The full report can be found using following link:
https://github.com/spacecraft-design-lab-2019/SolarActuators/blob/master/EnergyHarvesterBoardTest/Energy%20harvesting%20board%20testing%201112.pdf

The short summary is:
1. The charging circuit with diode will make 3.3 V stable but the EXT BAT cannot charge battery efficiently. 
2. With diode, the charging current is 14.7mA and we can only get 0.002 V battery voltage increase with 5 min charging. 
3. The Panda Sat solar cell cannot provide 3.3 V out independently under sunlight (get 1.3-1.5V).

klwall423, 2019-11-15:
Testing in sunlight: X gives power output of 0.8 V, -Y and Z give power output of 0.2 V. When measuring +3.3V and ground on the breakout board from the flat-flex cable (in late-afternoon sunlight), power output of 0.2 V on all 3 boards.

-Y and Z diodes output 1.6 V (as expected) across diodes, X output 0.7 V across diodes

maholli, 2019-11-16:
What is the output voltage of Vsolar? Not just across the diodes, but the actual voltage across the entire array of cells? This is the first thing to determine before chasing down the regulator/mppt circuit.

maholli, 2019-11-16:
## Where to measure Vsolar:

X panel:
[image]

-Y panel:
[image]

Z panel:
[image]

cwnaught, 2019-11-18:
Measuring between VSolar and Gnd in sunlight, we get:
Single Cell = ~1.7 V
Z = ~3.2 V
-Y = ~0.7 V
X = ~0.5 V

So it looks like the Z cell is performing properly. I think we misunderstood the alignment of the breakout board for previous tests - to fix this, always find the two adjacent pins that are shorted since those are the ground pins.

Also installed photosensor libraries on a raspberry Pi but uncertain if the power source for the sensors is on the board, if we have to supply the 3.3V from the Pi. Will ask at office hours today.

maholli, 2019-11-18:
@cwnaught photosensor is powered by 3.3V. The energy-harvesting circuit should provide this, but the sensor will not work if you aren't able to measure 3.3V between `3V3` and `GND`. 

As designed, the panels should be able to handle back-feeding (providing 3.3V from another source) onto the 3.3V rail.

cwnaught, 2019-11-19:
Yesterday, Max and I fed ~3.5V to the VSolar pad on the Z board and found that the energy harvester circuit was not working properly. Rather than outputting 3.3V, it output 0.01V (measured both from the breakout board and the 3.3V pad beneath the photosensor). The energy harvester chip was also very hot. We believe Z has an issue with the harvester circuit, and that X and Y have issues with both the cells and the harvester circuit.

Future work is going to be testing half-assembled boards, as well as comparing the board-energy harvester's response to the pre-made energy harvesters that we know work. 

Recommended procedure from here on out is to test the boards as we make them to more easily isolate issues.

thwhite, 2019-11-20:
We found that the cells on the Z board are not connected to the MPPT circuit. This is notably also the only board that actually works. We hypothesized that something about the MPPT circuit was impacting the solar power as well. 

To confirm, we unsoldered the diode on the X board, and found that afterwards, the solar circuit now provided 3.3V.

thwhite, 2019-11-20:
Now that we have the +Y board, we also tested that. It appears to have significant solder problems. Testing the connection, we found the same issues as the rest of the boards - everything seems correctly connected but no voltage overall. All of the voltages we did observe were substantially smaller than expected, which is quite likely a soldering issue.

thwhite, 2019-11-20:
More issues were found with the soldering of the +Y board - it appears that all of the solar cells are also backwards.

thwhite, 2019-11-20:
We desoldered the diodes on the board and it works now - provides correct voltage on all parts! The evidence continues that we suspect that the MPPT has troubles.

thwhite, 2019-11-23:
Working thread for documenting the burnwire testing: 
- No wire gauge below 24 fits into the holes we currently use. 
- Better to thread large lengths of the nicrome for a fuller cut.

thwhite, 2019-11-23:
Got the flat flex connector and PWM working. However, having trouble with continuity. Connectivity between flat flex connector and board is weak.

thwhite, 2019-11-23:
FIRST ISSUE FOUND: The batt-P net that starts at the flat-flex connector just doesn't connect to its other end on the burnwire circuit. No current flows. 

[image]

Dealt with this by greenwiring in a wire directly to power.

thwhite, 2019-11-23:
Second problem: 
Even with the PWM at 100% duty cycle, when power is confirmed supplied to the Batt-P net, GND to the GND net, and the PWM to the Burn-1 net, there appears to be no current flowing across burnwire circuit. 

As a sanity check, when Batt-P is shorted with VBURN, the nicrome cuts immediately.

thwhite, 2019-11-23:
Calling it for tonight, but summary - first issue seems to be an easily resolvable layout change. Second seems tough - need to look into the IC more to understand why it's not doing the thing, or perhaps what inputs it expects.

maholli, 2019-11-24:
> 
> 
> Calling it for tonight, but summary - first issue seems to be an easily resolvable layout change. Second seems tough - need to look into the IC more to understand why it's not doing the thing, or perhaps what inputs it expects.

Great detective work, @thwhite! You're correct, Batt_P was not properly connected. I also see why the PWM signal isn't working as intended:

The sort of MOSFET switch we're after will look something like this:
[image]

To make our circuit match the above diagram, we need to do the following:

1. Disconnect BURN1 from the pulldown resistor by cutting the trace between the via and the resistor. Make sure not to damage the trace on the other side of the via
2. Remove the two 10kohm resistors indicated below (easy with 0402 resistors. just get solder on the tip of your iron and lay it such that it touches both sides of the resistor. It'll heat up and you'll be able to scoot it off). 
3. Solder 1M ohm resistors in place of the two resistors you removed. 
4. Bridge solder between the pad you disconnected from the BURN1 via and adjacent leg of the MOSFET as shown below. 

[image]

klwall423, 2019-12-03:
Kill switches are working as expected

thwhite, 2019-12-03:
Note on the antenna test - since the antenna configuration has changed, mechanical is taking over this test and reorienting it towards just checking the new antenna config.

thwhite, 2019-12-03:
Acquired test data on how different gauges of Nicrome affect burnwire effectiveness! 

https://docs.google.com/spreadsheets/u/2/d/1ygNBy-Mw79r38jfn8FhKZXVq6TKu6gq1G2wAg_AYE3o/edit?usp=sharing_eip&ts=5de5fa76

thwhite, 2019-12-05:
Greenwiring mostly complete, but can't find 1MOhm resistors. Will ask around.

thwhite, 2019-12-05:
Update: greenwiring complete. Will return to the lab asap to test!

thwhite, 2019-12-06:
Update again: 

Having fully greenwired the circuit, the result is not encouraging. I observed no current flowing through the nicrome even at 100% duty cycle. At this time, the board passed continuity testing - the right things were connected to each other and there appeared to be no shorts. 

However, the board's integrity is failing from repeated hand-soldering. I can attempt to use the other Z board briefly, replicate the circuit in a breadboard, or take this result as good enough.

thwhite, 2019-12-06:
Next step - do it in a breadboard instead with a new IC so that we know for sure that nothing got fried in the greenwiring.

maholli, 2019-12-06:
Ok good news and bad news:
### Good news:

I green-wired as described in https://github.com/spacecraft-design-lab-2019/documentation/issues/132#issuecomment-557847274 and can properly pass current through the burn wire 🎉

### Bad news:

The current-carrying capacity of the flat flex cables are 0.5A. However, passing 0.5A through the burn wire for >10 seconds barely makes the burn wire warm. I slowly ramped up the current and found ~1.5A will burn the fishing line immediately (wasn't using the flat flex cable to supply the power, though).

maholli, 2019-12-07:
## Energy Harvesting Circuit Fix

looks like we soldered on one of the DNI resistors during assembly (sorry!!). Once I removed R15, the energy harvesting circuit seems to work. To reiterate, the resistors circled in red below should NOT be installed on the panel: 

[image]

zchen131, 2019-12-09:
The energy harvest board charging voltage plot is generated. 
Check: 
https://github.com/spacecraft-design-lab-2019/SolarActuators/blob/master/EnergyHarvesterBoardTest/Charging%20Voltage%20plot_2019-12-09-11-46-00.449.pdf
https://github.com/spacecraft-design-lab-2019/SolarActuators/blob/master/EnergyHarvesterBoardTest/Charging%20Voltage%20plot.PNG

As the figure shows, the EXT BAT provides 3.979V without connecting to the battery. After connected to the battery, energy harvest circuit is able to change the battery from 3.668V to 3.7V

zchen131, 2019-12-09:
Set input voltage to 3.0V by connecting 4 Diodes (100V 15A R-6) in series, EXT BAT before connects to the battery is 1.3V. After connecting the battery 3.17V. 
At the end of the test, the clamp which connected the battery with the meter loose so the voltage drop to 0V at the end.
In this mode, the V input is lower than V-output before connected to the battery. Thus the behavior should match with the yellow color plot in Fig 22 in the reference document (bq25570).
The test report can be found using following link:
https://github.com/spacecraft-design-lab-2019/SolarActuators/blob/master/EnergyHarvesterBoardTest/Vin_smaller%20than_Vout(EXT%20BAT)_2019-12-09-15-39-02.857.pdf

maholli, 2019-12-10:
> 
> 
> Set input voltage to 3.0V by connecting 4 Diodes (100V 15A R-6) in series, EXT BAT before connects to the battery is 1.3V. After connecting the battery 3.17V.

CORRECTION - you want to set the **solar input** to <3.3V NOT the external battery. This is because the energy harvesting circuit uses a boost regulator to charge the battery. Boost means it expects an input voltage less than the battery voltage (that is, when charging from low-impedance DC sources).

zchen131, 2019-12-10:
> CORRECTION - you want to set the **solar input** to <3.3V NOT the external battery. This is because the energy harvesting circuit uses a boost regulator to charge the battery. Boost means it expects [...]
Yes, the solar input is the input which is 3.3V. After connected solar input(i.e., 3V), the EXT BAT gives 1.3V as I state. After measure  EXT BAT, I connected it with the 3.7V battery which I have charged in the morning and measure EXT BAT which gives 3.17V. Then the EXT BAT stay around 3.17V and increasing slowly as The plot shows:  0.007 V increasing in 15 mins.

thwhite, 2019-12-11:
@maholli for the burnwire testing, what's the next move? I can start trying different combinations of burnwires to see if something works at 0.5A? (I also got us some test data from a different project with tests of different styles of burnwire if that would be helpful!)

zchen131, 2019-12-12:
The detailed energy harvest board voltage plot testing is available using the following link:
https://github.com/spacecraft-design-lab-2019/SolarActuators/blob/master/EnergyHarvesterBoardTest/Voltage%20plot_1211.pdf