spacecraft-design-lab-2019/documentation #142: OpenMV Proto1 Debug

NickG-NZ, 2019-11-12:
Testing Board without STM32 and Camera
- First test showed 0 volts from VCC. Determined it's because the Regulator needs it's enable pin pulled high. The enable pin is connected to the 10-pin flat flat flex connector and is intedned to be used by the main board.
- Vraw is 5V when powered by USB as expected

NickG-NZ, 2019-11-12:
Need to check what supply voltage OpenMV needs. Thought it was 3.3V but might not be. Also need to check STM32 supply voltage

NickG-NZ, 2019-11-17:
STM32 takes 3.3 V as expected. The OpenMV expects 3.6-5V as a raw input and regulates it to 3.3V. Testing has shown the board successfully outputs 3.3V to VCC. 
The board is also missing two resistor arrays (10K) because we didn't have the parts. These connect the SD card to VCC hence, the SD card won't work without them. Otherwise the board should function as expected

NickG-NZ, 2019-11-17:
Discovered that one of the missing resistor arrays RN3 is used by the MCU Boot0 and NRST Pins as well as the SD card. So the board likely won't work without it.

maholli, 2019-11-17:
Per https://github.com/spacecraft-design-lab-2019/documentation/issues/12#issuecomment-554794443, the STM32 from OpenMV-H7 and OV7725 from OpenMV3 have been installed on the camerav00 board. I also moved over the necessary resistor arrays and the RGB indicator LED for debugging purposes. 

## Next Steps

- [ ] need to power the board via USB, double-check 3.3V is correct, then work with @ashtofen to try and operate it as a normal openMV.
- [ ] 🥇ultimate test: is it possible to collect an image? Image is going to be blurry. Don't worry about putting a lens on just yet, just be careful with the board and don't set it down on anything abrasive.