spacecraft-design-lab-2019/documentation #188: ANSYS Thermal Modeling

Tdemuren15, 2020-01-24:
- Update material properties of components for greater accuracy

- Fine tune model to factor in radiation and conduction

thwhite, 2020-01-24:
Here are some thoughts I have:

Current status:

- Rough satellite model is complete! Need slightly more accurate material properties - I am uncertain, for example, about the heat capacity of lithium polymer batteries.
- Some mesh fixing to be done to make sure that body interfaces are being modeled right.
- BUT preliminary results exist!

Conclusion so far: the ANSYS model agrees with the final expected temperature of the body with the hand calculations. However, the modeling of the thermal transient involved suggests that there is non-zero hope that the outside structure of the satellite may insulate the batteries for some period of time. Key variable involved remains material properties, which are also the next large area of work.

Next steps:

- More accurate material properties.
- Better initial condition modeling - expected temperature equilibrium?
- Mesh resolution.

thwhite, 2020-01-24:
Other stuff to do:
- Set up transient heat flux to model whole orbit.

thwhite, 2020-01-24:
During meeting, also corrected for earth albedo - turns out that earth is incapable of reflecting sunlight while also blocking the sun at the same time. Heat flux reduced to ambient earth IR output value instead.

thwhite, 2020-01-27:
Increased mesh resolution. Redoing convergence study.

thwhite, 2020-01-27:
Found some more material data on FR-4 here: https://www.electronics-cooling.com/2002/08/simplified-transient-model-for-ic-packages/

thwhite, 2020-01-27:
Other next moves: remove a couple of unnecessary features from the part.

thwhite, 2020-01-27:
Goals coming out of Monday meeting with Manchester:
- Main priority is resolution over the whole orbit. Get the full transient done ASAP.
- Also add the new fancy parts. 
- Ok to model battery as lithium polymer.

Tdemuren15, 2020-01-27:
Addendum: Develop ode sim in Matlab to model the thermal conditions over the course of ~10 orbits

thwhite, 2020-01-29:
Docs on how to set up transient thermal conditions: https://www.clear.rice.edu/mech517/WB16/lectures_trainee/Mechanical_Intro_16.0_L10_Multistep_Analysis.pdf