spacecraft-design-lab-2019/documentation #192: Mechanical edits to boards

ElliseD, 2020-02-04:


klwall423, 2020-02-13:
Status Update:
1-4 completed on -Y board. Step file is now in -Y repository. Mounting hole radius has also increased from 1.1 mm to 1.3 mm
New updates to be completed:
-Add new antenna, resulting in killswitch movement

maholli, 2020-02-13:
@klwall423 please use the attached footprint for the new 2.3 GHz antenna. Rename file and remove the `.txt` in the filename. You can use a default KiCad antenna symbol like this: 
[image]. 

You should also add a pi-filter that will let us tune the frequency. Your schematic should look like this: 
[image]
We wont know the values for the inductor or capacitors yet. Please use 0201-sized parts for the inductor and capacitors:
[image]

Your layout should look like this, but you will have a different RF connector (thing on the far-left side):
[image]

For the antenna footprint, you will need to add a "keepout" region for planes as specific in the antenna datasheet. My keepout (on the bottom layer if the antenna is mounted to the top layer):
[image]

Let me know if you have questions. Let me know if you have questions. 

Antenna footprint:
[SXP.18.4.A.02.kicad_mod.txt](https://github.com/spacecraft-design-lab-2019/documentation/files/4201670/SXP.18.4.A.02.kicad_mod.txt)

klwall423, 2020-02-17:
-Y board should be up-to-date now with new antenna

westonb, 2020-02-18:
How many mm do we need to move up the solar pannels on the X board? Right now I have moved it up 2.25mm. With a bit more tinkering I could move it up another mm or so

ElliseD, 2020-02-18:
2.25mm will have the solar panels clear of the rails, thank you!

westonb, 2020-02-20:
I just updated the 3d models for Y- and it looks like the new 2.4GHz antenna is going to collide with the camera hole / one of the solar panels.  Are there any placement constraints on it? or can I move the solar panel up a bit and move the antenna right / down.

maholli, 2020-02-20:
Pic? @klwall423 and I did a quick sanity check with the datasheet when she first added the new patch. It should have silk screen outlining where it sits.

westonb, 2020-02-20:
[image]

Also, the matching network is currently in parallel with the antenna, not in series. I might need to move the antenna or the rf connector ever so slightly to get it to fit

ElliseD, 2020-02-20:
Importing the new -Y board, the new antenna overlaps the camera hole (maybe not a huge deal)/ one of the solar panels, as you said, but it also overlaps one of the screw holes, which has a keep-out area around it for the screw head to fit.  If the "L_Wuerth" components are going to be on the board when the satellite is assembled, they will need to be moved as they currently hit the battery donut.  In the screenshot, if those components are going to be on the board when the satellite is assembled, they will need to be moved so the Z board doesn't hit them.
[image]

The X boards look great.

westonb, 2020-02-23:
The footprint for the antenna was offset from the 3d model. I corrected it and moved the solar panel, it fits with only very slight overlap of the camera hole now. 

The "L_Wuerth" components are needed. Can you please provide some guidance on where I need to move them? I will also move the resistors you circled when I am doing that. 

Also, unless I am misinterpreting the 3d model of the flex receptacle, it seems like the antenna connector is going to collide with the flex. Where can I move it? 
[image]

Also, moved the diode on the x board. Please let me know if that fixed it.

maholli, 2020-02-23:
Yes the U.FL is going to need to be moved, as well as other fixes that I can see from that one image. See my notes below. 
@westonb please chime in if you've already editing the board. Otherwise...
- [ ] @klwall423 please fix the items in red below and then send it to Weston for review again.

westonb, 2020-02-23:
The RF matching network was previously not routed in the correct configuration so I did that last edit to the RF network. We are still moving things so I did not bother fixing the ground pour yet. Waiting to hear back from mechanical on where I can put the connector. 

 Do we have a fixed stackup? if so, what is the correct trace width for 50 ohms? I assume its a microstrip so you want some distance from the groundplane on the top layer.

maholli, 2020-02-23:
I hadn't bothered with a stack up yet. just cram the connector and matching network as close as you can to the antenna pin. Caps can move closer to the inductor on the pi filter IMO. I've measured the layout pictured in my [comment above](https://github.com/spacecraft-design-lab-2019/documentation/issues/192#issuecomment-586009266) and it does very well if you wanna match that.

westonb, 2020-02-23:
ok, I was trying to obey the keep-outs for the components but I will push the caps closer. 

Will make the changes when I hear back from @ElliseD about mechanical clearances.

ElliseD, 2020-02-23:
I added a dxf file to the electronics Slack for where the "L_Wuerth" components can go.  The X boards look good.